davidgfnet/superfw #6: AGB-001 stuck on white screen

TsilaAllaoui, 2025-05-11:
I know that some revision of Gba AGB-001 works great but i have tested on 3 different revision and none worked for me.
- AGB-CPU-02, AGB-CPU-03,AGB-CPU-10 : white screen after selecting rom with any patch, tried running pre patched rom but doesn't work either (no patching selected in menu), gb/c emulation work with some rom but have major visual glitch rendering it unplayable
- Other AGB revision: need to be tested because i don't own these
- NDS/Lite: works as intended
- Gameboy Micro: works as intended

So i think that we need a public document to fill for those who have working revision to find why it doesn't works as intended on AGB. Some say that older or newer revision works but i have AGB-CPU-10 which is one of the newer version and it doesn't works for me but work for some other. 

I rule out mod like IPS screen because i have modded AGS 101 screen AGB but doesn't works and unmodded vanilla other wich also doesn't works.

Found out too that some revisions have smaller EWRAM chip and some doesn't have AGB-REG chip at all. Maybe it's a hardware problem too. Or overclocking the crystal oscillator can help too.

Tried messing with waistate and timing related to writing to SDRAM and overclocking/de-overclocking EWRAM but no success.

Any help or advice is welcome. Thanks

TsilaAllaoui, 2025-05-28:
Update: only AGB-CPU-01 and 02 are affected. Those havze "CPU AGB" instead of "CPU AGB A" on the cpu SoC. I think that it's thz dmaCopy function that doesn't work for these revisions, reverting to writing directly to GBA_ROM address makes game boot instead of white screen but game will have visual glitch and crasj after some point

xCastorm, 2025-06-06:
I'm confused, you're saying this doesn't work on AGB-001 Gameboy Advance? I have one and it works fine?

TsilaAllaoui, 2025-06-06:
@xCastorm it's because you are using AGB-CPU-3 or AGB-CPU-4 or AGB-CPU-10 (32pin screen). Only AGB-CPU-02 and lower will not work. @davidgfnet already have idea how to fix it, waiting for new build

the-real-mcarn, 2026-01-06:
@chrisf95 +1
Same issue here, AGB-CPU-10 with 32 pin screen. Would boot to black or with glitchy audio with any game. Also solved it by disabling fast ROM loading. Same supercard works perfectly fine in a GBA SP without changing the settings.