davidgfnet/superfw #10: Mother 3 splash screen issue

seeseemelk, 2025-05-17:
When running Mother 3 (fan translated), the second splash screen looks completely garbled when running of the Supercard SD.

I'm not yet sure why this is, my best guess at the moment is that some video register is not correctly being reset when the game starts.

I've tried it with the following settings:
 - Skip BIOS Boot + patching enabled + Fast Rom Loading enabled (image A)
 - BIOS Boot + patching enabled + Fast Rom Loading disabled (image B)
 - BIOS Boot + patching disabled + Fast Rom Loading disabled (doesn't even start)

# Image A
[image]

# Image B
[image]

# Expected
[image]

I am at this point in time not 100% certain if this is related to a bug in this firmware, or if it's a hardware issue with my GBA/Supercard SD. Given that most other games seem to run fine, I'm steered towards thinking the former.

samfadel, 2025-05-17:
What is superfw version u use?

seeseemelk, 2025-05-17:
Right now v0.16, though I also encountered the first image in v0.15. I hadn't tried messing with the settings in v0.15

seeseemelk, 2025-05-17:
With the original supercard sd firmware, the splash screen shows as expected. So it must be a bug with the superfw firmware.

seeseemelk, 2025-05-17:
So, turns out this was an issue with the built in patch database. After generating patches instead, it works as expected

closed 2025-05-17 by seeseemelk

samfadel, 2025-05-20:
I tested and it works also it saves without any problem.
Check your sd card

davidgfnet, 2025-05-25:
Please report game (and patches :P) issues directly at: https://compat-superfw.davidgf.net/

That should help me fix any patch generation bugs. Thanks!